saidlm/Domoticz-Solax-plugin #3: Call to function 'onHeartbeat' failed error

averter, 2024-05-17:
Hello,

Thanks for this amazing plugin! I am using a X1-Hybrid-G4 solax inverter and am unable to connect to it for some reason (debug log below). Note: I had to remove one line containing private information about the network/passwords.
If you can give me any pointers or other advice I'd be really grateful and/or let me know if you need any other information.

Best regards

````
2024-05-17 15:54:40.050 X1-Hybrid-G4 : Pushing 'onHeartbeatCallback' on to queue
2024-05-17 15:54:40.101 X1-Hybrid-G4 : Processing 'onHeartbeatCallback' message
2024-05-17 15:54:40.101 X1-Hybrid-G4 : Acquiring GIL for 'onHeartbeatCallback'
2024-05-17 15:54:40.101 X1-Hybrid-G4 : Calling message handler 'onHeartbeat' on 'module' type object.
2024-05-17 15:54:40.101 X1-Hybrid-G4 : onHeartbeat
2024-05-17 15:54:40.101 X1-Hybrid-G4 : Connecting to: 192.168.1.103:502
2024-05-17 15:54:40.289 X1-Hybrid-G4 : Successful attempt number: 1
2024-05-17 15:54:40.290 X1-Hybrid-G4 : Local context:
2024-05-17 15:54:40.290 X1-Hybrid-G4 : ----> 'Devices' 
2024-05-17 15:54:40.290 X1-Hybrid-G4 : ----> 'Domoticz' '<module 'Domoticz' (built-in)>'
2024-05-17 15:54:40.290 X1-Hybrid-G4 : ----> 'Images' '{}'
2024-05-17 15:54:40.290 X1-Hybrid-G4 : ----> '_plugin' '<plugin.BasePlugin object at 0x7fff79ee1850>'
2024-05-17 15:54:40.290 X1-Hybrid-G4 : ----> 'time' '<module 'time' (built-in)>'
2024-05-17 15:54:40.290 X1-Hybrid-G4 : Acquiring GIL for 'onHeartbeatCallback'
2024-05-17 15:54:40.289 Error: X1-Hybrid-G4 : Call to function 'onHeartbeat' failed, exception details:
2024-05-17 15:54:40.290 Error: X1-Hybrid-G4 : Traceback (most recent call last):
2024-05-17 15:54:40.290 Error: X1-Hybrid-G4 : File "/home/pi/domoticz/plugins/Solax/plugin.py", line 268, in onHeartbeat
2024-05-17 15:54:40.290 Error: X1-Hybrid-G4 : _plugin.onHeartbeat()
2024-05-17 15:54:40.290 Error: X1-Hybrid-G4 : File "/home/pi/domoticz/plugins/Solax/plugin.py", line 118, in onHeartbeat
2024-05-17 15:54:40.290 Error: X1-Hybrid-G4 : self.updateDevices(inputRegisters)
````

averter, 2024-05-17:
Solution:
Replace
```python
    def updateDevices(self, registers):
        decoder = BinaryPayloadDecoder.fromRegisters(registers, byteorder=Endian.BIG, wordorder=Endian.LITTLE)
```
by
```python
    def updateDevices(self, registers):
        decoder = BinaryPayloadDecoder.fromRegisters(registers, byteorder=Endian.Big, wordorder=Endian.Little)
```
:-)

saidlm, 2024-05-20:
Thank you.

Martin

saidlm, 2024-06-03:
It is not so easy unfortunately. The fix sent by Averter doesn't work to me. I am just install it on test environment and have to change it back to BIG and LITTLE.

It looks to me there is some dependency to OS or libs. 

@Averter, can you send me some details about your environment, please? OS, Domoticz version, pymodbus version, etc. Thank you.

Martin

saidlm, 2024-06-03:
Looking at https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/payload.py and https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/constants.py it looks to me that BIG and LITTLE are the corrects ones

averter, 2024-06-03:
@saidlm sure. 
Hardware: Raspberry pi 5.
OS: Linux 6.6.28+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux
Domoticz version: 2024.4
Pymodbus version: 3.0.0-7
Python version: 3.11.2

averter, 2024-06-03:
> Looking at https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/payload.py and https://github.com/pymodbus-dev/pymodbus/blob/dev/pymodbus/constants.py it looks to me that BIG and LITTLE are the corrects ones

I think that it's because I've installed pymodbus from the distro package, which is a quite old version, 3.0.0-7, according to [this webpage](https://packages.debian.org/search?keywords=pymodbus&searchon=names&suite=stable&section=all).

saidlm, 2024-06-03:
My one is even older 2.5.2. I am just upgrading my Domoticz docker image to be able to work with Bookworm now.

saidlm, 2024-06-17:
I tested different modbus modules and I was not able to replicate the issue. For me it always works with BIG and LITTLE. 

Martin

closed 2024-06-17 by saidlm

averter, 2024-08-11:
@saidlm it has to do with pymodbus version. If the version starts with 3.3 or 3.0 it ought to be lowercase `big` and `little`, otherwise it needs to be uppercase `BIG` and `LITTLE`. See an implementation [line in HA code](https://github.com/wills106/homeassistant-solax-modbus/blob/4647ba453a4f088b66b8202f4ace42b8baad005f/custom_components/solax_modbus/__init__.py#L43) 

I leave it up to you to decide if you want to include such condition in your plugin. I can do a new PR (let me know).

saidlm, 2024-08-15:
@averter thank you for the offer. I am not sure if it make sense to spend time on that. Anyway, if you would like to do it I am happy to merge it in to the code.

Thank you for cooperation.

Martin

infoedkok, 2024-09-09:
I'm investigating a problem with the OnHeartBeat call in another project, https://github.com/nmakel/solaredge_modbus/issues/96

There is a randomly "huge" timegap between the OnHeartBeat call and its processing, in my case of almost half a minute! 

Where a "regular" loop is proccessing in milliseconds after the OnHeartBeat call.

(SolarEdgeModbus plugn for Domoticz in a docker container on Synology)

What could cause such a huge delay?

saidlm, 2024-09-10:
Hi,
I haven't have such experience. My setup is very similar. I have been also using Docker. The difference is environment below. I have RPi and Raspbian on it. BTW my Domoticz Docker stack is published here: [https://github.com/saidlm/Domoticz_stack](https://github.com/saidlm/Domoticz_stack).

Plugin functions are called from Domoticz core. I haven't spent any time to review the code. I can't help you with that. You can try to contact Domoticz developers directly. Hopefully they will be able to help you with that.

In theory there can be issue with RTC cause by your setup. I solved some issue with precision of time in Docker containers but it was about milliseconds not minutes.

What about CPU load on your Synology? Or processes priority?

Martin