saidlm/Domoticz-Solax-plugin #8: Question modbus address

jfh75, 2024-08-02:
Hello Martin,

I am very interested in you plugin. Could you please tell me what line is the modbus address of the inverter please ? 
Because I think this is the reason why it is not working here. My solax address is 3, not 1.

Thank you for you kind help,

Best regards,

JFH

saidlm, 2024-08-02:
I have just added Unit ID as a standard parameter of the plugin. Please try it and let me know if it helps.

Martin

jfh75, 2024-08-02:
Oh no, in the end that was not the problem.
Though I see when connecting to the N510 : CONNECTED (2). So it connects but gets wrong answer I believe.
Here are domoticz logs : 

2024-08-02 13:29:29.021 solax G4 Hybrid: Pushing 'onHeartbeatCallback' on to queue
2024-08-02 13:29:29.072 solax G4 Hybrid: Processing 'onHeartbeatCallback' message
2024-08-02 13:29:29.072 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:29.072 solax G4 Hybrid: Calling message handler 'onHeartbeat' on 'module' type object.
2024-08-02 13:29:29.072 solax G4 Hybrid: onHeartbeat
2024-08-02 13:29:29.072 solax G4 Hybrid: Connecting to: 192.168.0.9:502
2024-08-02 13:29:32.074 solax G4 Hybrid: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2024-08-02 13:29:32.074 solax G4 Hybrid: Unable to read input registers.
2024-08-02 13:29:32.074 solax G4 Hybrid: Updating devices from Local array
2024-08-02 13:29:32.074 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:39.001 solax G4 Hybrid: Pushing 'onHeartbeatCallback' on to queue
2024-08-02 13:29:39.051 solax G4 Hybrid: Processing 'onHeartbeatCallback' message
2024-08-02 13:29:39.052 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:39.052 solax G4 Hybrid: Calling message handler 'onHeartbeat' on 'module' type object.
2024-08-02 13:29:39.052 solax G4 Hybrid: onHeartbeat
2024-08-02 13:29:39.052 solax G4 Hybrid: Connecting to: 192.168.0.9:502
2024-08-02 13:29:42.056 solax G4 Hybrid: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2024-08-02 13:29:42.057 solax G4 Hybrid: Unable to read input registers.
2024-08-02 13:29:42.057 solax G4 Hybrid: Updating devices from Local array
2024-08-02 13:29:42.057 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:49.034 solax G4 Hybrid: Pushing 'onHeartbeatCallback' on to queue
2024-08-02 13:29:49.085 solax G4 Hybrid: Processing 'onHeartbeatCallback' message
2024-08-02 13:29:49.085 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:49.085 solax G4 Hybrid: Calling message handler 'onHeartbeat' on 'module' type object.
2024-08-02 13:29:49.085 solax G4 Hybrid: onHeartbeat
2024-08-02 13:29:49.085 solax G4 Hybrid: Connecting to: 192.168.0.9:502
2024-08-02 13:29:52.090 solax G4 Hybrid: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2024-08-02 13:29:52.090 solax G4 Hybrid: Unable to read input registers.
2024-08-02 13:29:52.090 solax G4 Hybrid: Updating devices from Local array
2024-08-02 13:29:52.090 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:59.024 solax G4 Hybrid: Pushing 'onHeartbeatCallback' on to queue
2024-08-02 13:29:59.074 solax G4 Hybrid: Processing 'onHeartbeatCallback' message
2024-08-02 13:29:59.074 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:29:59.074 solax G4 Hybrid: Calling message handler 'onHeartbeat' on 'module' type object.
2024-08-02 13:29:59.074 solax G4 Hybrid: onHeartbeat
2024-08-02 13:29:59.074 solax G4 Hybrid: Connecting to: 192.168.0.9:502
2024-08-02 13:30:02.078 solax G4 Hybrid: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2024-08-02 13:30:02.079 solax G4 Hybrid: Unable to read input registers.
2024-08-02 13:30:02.079 solax G4 Hybrid: Updating devices from Local array
2024-08-02 13:30:02.079 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:30:09.005 solax G4 Hybrid: Pushing 'onHeartbeatCallback' on to queue
2024-08-02 13:30:09.055 solax G4 Hybrid: Processing 'onHeartbeatCallback' message
2024-08-02 13:30:09.055 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:30:09.055 solax G4 Hybrid: Calling message handler 'onHeartbeat' on 'module' type object.
2024-08-02 13:30:09.056 solax G4 Hybrid: onHeartbeat
2024-08-02 13:30:09.056 solax G4 Hybrid: Connecting to: 192.168.0.9:502
2024-08-02 13:30:12.060 solax G4 Hybrid: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2024-08-02 13:30:12.060 solax G4 Hybrid: Unable to read input registers.
2024-08-02 13:30:12.060 solax G4 Hybrid: Updating devices from Local array
2024-08-02 13:30:12.061 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:30:19.039 solax G4 Hybrid: Pushing 'onHeartbeatCallback' on to queue
2024-08-02 13:30:19.090 solax G4 Hybrid: Processing 'onHeartbeatCallback' message
2024-08-02 13:30:19.090 solax G4 Hybrid: Acquiring GIL for 'onHeartbeatCallback'
2024-08-02 13:30:19.090 solax G4 Hybrid: Calling message handler 'onHeartbeat' on 'module' type object.

Thank you for your help,

JFH

saidlm, 2024-08-02:
In fact inverter itself doesn't provide Mod-bus over TCP. There is converter for it anyway. The difference is in the model. I am using Solax one which is USB dongle with WiFi in my case. 
The question is which functions are supported and how it is implemented in your one.

I see two possible root causes of your issue:
1) Problem with timing on Mod-bus. I have fond this commend in documentation: "The query cycle is expected to be controlled at about 1 second." May be there is some problem in the reading via RS-485 in terms of timeouts.
2) For reading all the data from the inverter I am using possibility to read more value in one reading cycles. In one of early version I read up to 200 registres in one read. Finally I recognized it is not stable. Sometimes it produce input/output errors and sometimes it caused that dongle stop to answer ModBus at all. In current version the function is different. There is loop which repeats more read cycles with smaller amount of data. It looks to me stable now ...but is valid for original Solax dongle.  
Try to change the number of bytes in one cycle. It is in function updateDevices. Look for "inputRegisters = self.getInputRegisters(0, 290, 10)". Currently it reads 100 registers and request 10 registers per one red function.

Hopefully it helps.

Martin

averter, 2024-08-10:
I am having exactly the same issue as @jfh75, `Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)`, although I am connected over the com/lcd port with a [waveshare RS485 to RJ45 Ethernet Module](https://thepihut.com/products/industrial-serial-server-rs485-to-rj45-ethernet?variant=41374103666883) 

@saidlm So should we try to increase/decrease the number of registers in the getInputRegisters, as for example using `inputRegisters = self.getInputRegisters(0, 290, 20)`? What range of values can we work with? Can you share some relevant modbus documentation? In my experience the solax dongle is unreliable, specially if the inverter is to be used for a performance critical operation, such as charging an EV in a solar mode, as per [my separate post](https://github.com/saidlm/Domoticz-Solax-plugin/issues/7). I'm happy to report here my findings with the waveshare but I need to have something to work with. Please can you spend some time giving me some guidance? Thanks.

averter, 2024-08-11:
Really tiring to learn how modbus works...hats down to you guys

I've installed openmodscan and have successfully been able to connect with the waveshare using a length (number of registers read) of 10 as you have defined in the plugin
[image]

However the client.read_input_registers on [this line](https://github.com/saidlm/Domoticz-Solax-plugin/blob/main/plugin.py#L444) always fails with the above error, no matter what combination of steps that I use (10, 1, 20, 50, 100)... I've also tried a start address of 1 instead of 0, still no luck...

averter, 2024-08-12:
Hi, I've just created a [PR](https://github.com/saidlm/Domoticz-Solax-plugin/pull/9) which should fix this issue. Let me know if it works. I had to fiddle around quite a bit until I finally found [this comment](https://github.com/pymodbus-dev/pymodbus/issues/696#issuecomment-1001351248) which did the trick.

saidlm, 2024-08-12:
Thanks,
Martin

closed 2024-08-12 by saidlm