saidlm/Domoticz-Solax-plugin #10: SoC mode set to charge battery actually discharges it

averter, 2024-08-16:
Hello,

Today I've tried to charge the battery during night using the new remote control functionality, specifically SoC mode 3. The controls were set to charge up to 75% in a maximum time of 3000s and a maximum power of 1000W
[image]
Next I've started the mode, and as you can see in the above picture the actual values seem to be correct. 
[image]
However, what actually happened is that the inverter started sending 3000W from the battery to the grid...
[image]
My guess is that the remote control timeout of 3000s is mixed up with the control charge/discharge power, which then is set at 3000W, and the SoC target might be mixed with one of the other variables set at 0 (Power/Energy Target or Duration time) and therefore it tries to reach a fully discharged state.

averter, 2024-08-16:
@saidlm 
This is the registers table 
[image]
 and your builder where you add each of the control settings is shown below
````python
        Domoticz.Debug("Starting ModBus Remote Control.")
        
        builder = BinaryPayloadBuilder(byteorder=Endian.BIG, wordorder=Endian.LITTLE)
        modes = (0, 1, 2, 3, 7)
        mode = modes[int(int(self.__RC_SETTINGS['Mode'])/10)]
        
        builder.reset()
        builder.add_16bit_uint(mode)                                            # Remote Control Mode
        builder.add_16bit_uint(1)                                               # TargetSet type = SET
        builder.add_32bit_int(int(self.__RC_SETTINGS['PowerTarget']))           # Target Active Power
        builder.add_32bit_int(0)                                                # Target Reactive Power
        builder.add_16bit_uint(int(self.__RC_SETTINGS['DurationTime']))         # Time of Duration
        builder.add_16bit_uint(int(self.__RC_SETTINGS['SOCTarget']))            # Target SOC
        builder.add_32bit_uint(int(self.__RC_SETTINGS['EnergyTarget']))         # Target Energy
        builder.add_32bit_int(int(self.__RC_SETTINGS['ChargerPower']))          # Charge / Discharge Power
        builder.add_16bit_uint(int(self.__RC_SETTINGS['TimeOut']))              # Remote Control Timeout

        payload = builder.to_registers()
````
and although the number of bits seems right the order does not match the above table at all. For example, Target Energy should come before the Target SOC, and all of the bits of "actual" values below the mode are not accounted for.

saidlm, 2024-08-17:
Hi,
there are two different range of registers for writing and reading. The writing uses mod-bus function 0x10 - Write multiple register.
[image]
The reading of current state of the inverter is done via mod-bus function 0x04 - Read input register. It is the part you have mentioned above. Domoticz shows you real actual values which were read from the inverter. There are two reason for that. a) It is to allow you to check that it uses exactly what you want. b) If there is another master device which makes any changes you can see the real state of the inverter.

Strange thing is that it looks ok. You can see in Input registers exactly what you set in Multiple register up. There is no automation logic in the plugin. My idea is to create just interface to Inverter and let user to do their automation in event scripts. I have just reviewed builder and decoder as well and I haven't found any problem there. There is only small issue with type int vs uint for reading Target SoC from inverter but it can't influence the behavior of the inverter.

Martin

saidlm, 2024-08-17:
I am just running the test and it seems to me it is working properly.
[image]

M.

averter, 2024-08-17:
@saidlm Just to make sure that we followed the same steps, you:

1. Set the target soc, the charge/discharge power and the control timeout
2. set mode 3 (soc) in the control mode switch
3. and turned the mode on
Right?

saidlm, 2024-08-17:
Yes, right.

M.

BTW. What is the version of your inverter software?

averter, 2024-08-17:
I hope this is the info you're after:

DSP: 2.07
ARM: 2.03

averter, 2024-08-27:
@saidlm 
I've tried to charge the battery again, this time setting different values for the variables, to see if I could get it to work as it should
[image]
but exactly the same happened. The battery started discharging at 3000W into the grid (despite the now 600s in remote control timeout (a coincidence)). 
Do you think that it might be due to the version of the inverter?
The debug log is also not helpful. Are there any variables worth adding to the log so that we can understand what's wrong?
Any other things worth testing?
I really want to get this functionality of the plugin working before winter, for obvious reasons :-)

averter, 2024-08-27:
Yess, it's probably because of the inverter's firmware version... Version 3.29 was the latest version that I could find, and the registers look quite different than the ones you've shown above.
[Solax-Hybrid X1&X3-G3 ModbusTCP&RTU V3.29- English.pdf](https://github.com/user-attachments/files/16760133/Solax-Hybrid.X1.X3-G3.ModbusTCP.RTU.V3.29-.English.pdf)
This is the new table

saidlm, 2024-08-28:
It looks to me for different model. I have G4 and the plugin is written for G4. The documents which I have uploaded to doc folder are also for G4.  V3.24 is the latest one which I have found.

saidlm, 2024-08-28:
Back to the software version which I am running on it. My Internal code is now 43 09 42. It includes versions v1.43 is for DSP and 1.42 is for ARM. I hope it is the last version. It was updated by my supplier about 10 day ago.

BTW. I have found this link: https://app.box.com/s/3nvo7ic523fhojf8uuto105q9a8dgk9n/folder/79352264617
There is collection of different software for Solax devices. Honestly speaking I am not about to test it. I would like to be sure that the version which I am running is the official one.

Martin

averter, 2024-08-29:
The above values were read from the "About" menu of the inverter. Via the wifi dongle system tab it shows a firmware version of 3.001.02. Is this number related to the wifi dongle or the inverter? I'm getting mixed up with so many versions.

[image]

averter, 2024-08-31:
The v3.24 document is quite clear in how the versions are structured (table below)
[image]
However the v3.29 and v3.28 which I was also able to find online don't provide such table. So I can't find ANY document that specifies the registers for the firmware version (2.07/2.03 DSP/ARM) that is installed in my inverter :-(. Worst case scenario I'll have to contact Solax directly for this information... Before doing this, do you know if there is a collection of solax modbus registers documentation for different versions, similar to what you mentioned before for the firmware of solax devices? Thanks.

saidlm, 2024-09-01:
Honestly speaking I don't know exactly what is software version. According to information from the link software version is 1.4X now. The same information I have from my supplier. And it corresponds with internal codes. 
The versions which you are talking about are something different I guess. I am not sure but I guess it is HW version of the board.

Martin

saidlm, 2024-09-01:
I have look for changelogs etc. I have also asked my supplier for that but have no success :-(. The answer is that they are not allowed to share this kind of information.
Only what I have from Solax/my supplier is list of Mod-Bus registers v 3.21. Everything else has been found somewhere on the Internet.

Martin

averter, 2024-09-01:
@saidlm Cheers. I've contacted solax regarding this issue and will let you know if they send me anything useful. If it is a version-related problem I can always update the firmware to the latest version, which hopefully will be 1.4X as yours.
Off-topic: Besides mode 3 "SOC mode", I am also interested in mode 7 "Self-consume - Charge Only mode" to charge my EV rather than the battery during night in Winter. Since when in this mode the battery cannot be discharged and can only be charged with PV, this mode will effectively pull energy from the grid to charge the EV. 
[image]

In the remote control this is labelled as "Charge Only" I believe. I will test to see if there is any difference relatively to mode 3, although it is unlikely...

averter, 2024-09-03:
> Hi, there are two different range of registers for writing and reading. The writing uses mod-bus function 0x10 - Write multiple register. [...]

@saidlm I was able to get confirmation from Solax that I should use v3.21 for this inverter. However, when I look through this document there is no table with the parameters as the one that you have shown above. Actually, there is no table concerning "write multiple registers" at all (only read registers and single write register)! I assume that you have used v3.24 instead? I can ask to Solax what should the write multiple registers be, although if you also have a X1-Hybrid-G4 it shouldn't matter...

saidlm, 2024-09-03:
Try to ask your supplier or Solax directly for the upgrade of your firmware. They can do it remotely. I hope it will solve the issue. Unfortunately multiple registers is only way how to control the inverter. Any other possibility is using EEPROM which is wrong way.

Martin

averter, 2024-09-03:
This has been moved to my top priority since the sun has decided to go on holidays now in September...
I think you are right, it's most likely because of the outdated firmware. Only in v3.22 has the write multiple registers table been added
[image]

So if my firmware is an older version it won't be compatible with such functionality. I'll ask for an update and see if then works out.

averter, 2024-09-03:
I'm happy to report some progress. My firmware is now updated to DSP v1.45 and ARM v1.42.
I've tested charging the battery over night using mode 3 (SOC) and now it does not starts discharging at maximum power, but the battery is also NOT being charged. Instead, the battery switches off, and all the energy starts coming from the grid, i.e., it does not try to mach the target SoC (75% in my case). 
[image]

[image]

@saidlm Any thoughts? Thanks for the continued support.

averter, 2024-09-05:
Hi Martin. Another update. Solax have come back to me with the most up-to-date document v3.34 compatible with the inverter (attached)
[(Solax)Hybrid X1&X3-G4 ModbusTCP&RTU V3.34-English_240313 .pdf](https://github.com/user-attachments/files/16886104/Solax.Hybrid.X1.X3-G4.ModbusTCP.RTU.V3.34-English_240313.pdf)

The write multiple register table is as follows
[image]
[image]
but this seems nearly identical to the one you've used (previous comment above). The only exception is the 0x0084 (Target energy field) which also includes 0x0085? :-S

saidlm, 2024-09-05:
It is awesome. I have already uploaded it into doc folder. 
The address 0x0085 is in fact present in both documents. There is UINT32 which uses 2 regs. It means it is covered by 0x0084.
The main thing which makes me happy is information about access to data from Solax EV Charger. I have already tested it and it looks to me that it works. I have a plan to add it into the plugin as optional feature.
Unfortunately it refers to version 3.3. of the EV Charger Mod-Bus document and have only version 2.8.  I have received it from my supplier as the last version :-(.  It better than nothing, yes.

Many thanks for the document!

Martin

averter, 2024-09-06:
Well...I'm happy that you're happy, but my problems just keep getting worse... Today my battery dropped to 10% during night!!!
As mentioned I have upgraded to the latest version and am still not able to charge the battery from the grid, even though my inverter is the same as yours. Why? What else can I test? Please help!

averter, 2024-09-07:
Things that I have/am testing:

- To rule out communication issues due to the use of the waveshare I have tried to use the wifi dongle again (I assume that is what you are using). Results were exactly the same.

averter, 2024-09-07:
To attempt to rule out any issues with the form of the payload I've added the following debug line to the code
````python
        Domoticz.Debug(f"Ready to write multiple registers at 0x: {payload}")
```
and this is the output (which seems correct) for when I try soc mode to charge up to 100% at max 2000W in 600s
````
2024-09-07 12:20:54.492 Solax: Ready to write multiple registers at 0x: [3, 1, 0, 0, 0, 0, 0, 100, 0, 0, 2000, 0, 600] 
````

saidlm, 2024-09-09:
Hi,
fist of all the information about my setup. I have my inverter in **Backup mode** with default settings. It can be reason why the behavior is different.
The second thing is - I have never tested it with such long interval. All my tests use intervals in minutes not more. The question is also implementation and stability of these functions in Solax software. I have read that there is issue with long intervals. The information was about one year old but ...
The third thing is the mode. I am focusing on the first one - Power. I have spent more time on tests of this mode and only small amount of time on the others. I will try to start similar tests as you and will see what will happen.

Martin

averter, 2024-09-09:
> Hi, fist of all the information about my setup. I have my inverter in **Backup mode** with default settings. It can be reason why the behavior is different. The second thing is - I have never tested [...]

Many thanks Martin. When you say interval do you mean the `RemoteControlTimeout`? I've been using 600s=10min which I thought would be fine - let me know if I should drop it to a lower value. I can also try to run tests in **Backup mode** and see if it works (thanks for sharing your setup).

At least I have been able to make the inverter charge the battery during night when energy is cheap, and so I am safe for now, although I'd like to be able to automate it using your plugin. It's worth stating that it was quite difficult to configure the inverter to charge in **Self mode** during night. I had to change the options of the mode to: 
1. Enable charging from the grid
2. Set the force charge period start and end
3. Charge battery to 90%

Only after all three of the above were configured did it start charging during night.

saidlm, 2024-09-09:
Hi,
I have just finished one of the tests. It started from 86% SoC about 300W from PV (cloudy today ;-)) and about 200W load. I tested SoC mode, 95% requested SoC, 3000W charge power and 900s Timeout.
It worked almost as expected. Finish flag and Mode changed just after I pushed trigger. Inverter power was close to 3000W while it charged. After about 10 minutes it was at 95% and finished charging. Finish flag went up as expected. 
Up to this time is exactly what I have expected. Only difference is Mode. There was Remote SoC up to Timeout i finished. My expectation was that Finish flag and Mode is changed as soon as Timeout is finished OR requested SoC is reached.

Well it looks to me that in general it works properly.

Martin

averter, 2024-09-09:
If the only difference is the mode, then I'll try with Backup mode instead of self-use and see if it works.

saidlm, 2024-09-09:
Look carefully at this part from Solax Academy document:
_Exit logic: if the SOC target value (0x0083) is not updated within the set time (0x0088) after completing the commands, then this mode exits. **When the SOC target value is not reached, the mode runs until the next command is entered.**_

May be it is the answer to same questions.

Martin