saidlm/Domoticz-Solax-plugin #11: New Version Plugin Solax

tracteur12, 2024-09-08:
Hello,

Thanks a lot for your plugin.

I'm trying to use the new version, but I can't to test it. I have to add number in "Inverter ModBus ID" (Default is 1), In Last version this information was not asked.
I 'm trying different number but it don't work, 

[image]
[image]

Where I can found "Inverter ModBus ID"? or maybe, if you check my error list you find the problem.

Thank.

Romain

saidlm, 2024-09-09:
Hi,
are you sure you have properly installed pymodbus library? Based on log I am afraid that library is missing on you system or has not been properly installed.
My recommendation is to install it using command **pip3 install -U pymodbus** instead of install system packages such as apt for Debian. The reason for it is the version of the library.

Back to your original question. If you have solax WiFi communication dongle leave ModBus ID as it is - "1". In case of RS485 communication via RS845/Mod-bus <-> Ethernet adapter you have to set proper ID in Inverter settings and the same in the plugin.

Martin

tracteur12, 2024-09-10:
Hi,

Ok, I'm updated to debian 12 and I had problem with "externally-managed-environment" when I m tryed to insall "pip3 install -U pymodbus". Finaly, I'm stayng in debian 11 is more stable. Now It works. My apologize. Thanks. Romain.

saidlm, 2024-09-11:
Hi,
There are two possible ways how to deal with "externally-managed-environment" .
1) You can install the module under same user which is used for running Domoticz
2) You can switch off the functionality by the command: rm /usr/lib/python3.11/EXTERNALLY-MANAGED
For my Docker image I have chosen the second option. It works to me properly.

Martin

tracteur12, 2024-09-12:
Hi,
Thank for you information and I test to do it.

I try to find some information with "onheartbean" but I didn't find answer. Have got any information?

Cdlt,

Romain
[image]

saidlm, 2024-09-19:
Please try to use the last version of the plugin. 

Martin

averter, 2024-09-21:
@tracteur12 I have to say that the plugin is very picky with the pymodbus version that you use, and there are lots of threads online with issues related with communication via pymodbus. I had to install pymodbus version 3.6.8 for the plugin to work correctly; a version more recent than 3.6.8 did not work. Specifically, I've installed it on my raspberry pi 5 using the following command:
`sudo pip3 install -U --break-system-packages pymodbus==3.6.8`

tracteur12, 2024-09-22:
Hi,
Thank you for the support. He work know (Pi3+ debian 12).
Romain.

saidlm, 2024-10-01:
Hi,
unfortunately the API of the pymodbus library is changing a lot :-(. Hopefully the last version of plugin is dealing little bit better with that. 

Martin

averter, 2024-10-01:
@tracteur12 If it works then close the issue?

closed 2024-10-06 by tracteur12