saidlm/Domoticz-Solax-plugin #12: Mode 7 Self-Consume - Charge Only Mode stops working after a couple of minutes and reverts back to original mode

averter, 2024-09-21:
Hello,

Today night I have tested `Mode 7: Self-Consume - Charge Only Mode` to be able to charge my EV from the grid without consumption from the battery. Interestingly, the mode starts working fine, but after 3-5 minutes it reverts back to the original self-use mode that I have by default, and starts consuming from the battery :-O. I have enabled the mode once at 00:31:04 and then again at 01:26:28, and it stopped on its own at 00:34:36 and 01:31:29, respectively. See the log below 
[image]
The following "Disabled" state it's me shutting down the mode. According to the [description of the mode](https://kb.solaxpower.com/solution/detail/2c9fa4148ecd09eb018edf67a87b01d2) there is no timeout variable for this mode, i.e., it should keep active until it is disabled, correct?

saidlm, 2024-10-01:
Hi,
unfortunately I can't do anything with that. It is not feature of the plugin but it is the behavior of the inverter itself. The plugin doesn't include any automation function it is only interface between Domoticz and Inverter. 
Unfortunately you are not only person who has problem with that. I have read many similar issues in HA forum. 
The solution is to use only sort intervals and repeat them many times. In HA it is part of the plugin to "hide" the problem. From my point of view it is not good idea. I think it is better to have plugin only as an interface to inverter and everything else is end users' hands. It is easy to write event script which will provide functionality to fulfill exact requests.

Martin

averter, 2024-10-01:
I see...So basically I could change my dzvents script which charges the EV at night, so that if the `Solax - Remote Control Mode` device changes back to "self consume" before the EV is charged it reactivates the "self consume - charge only" mode. Sort of as a watchdog system? Is this what you're suggesting?

averter, 2024-10-15:
I've managed to develop and test a dzvents script that is triggered whenever the inverter gets back to self-consume mode and re-activates charge-only (shown below)
```` lua
return {
	on = {
	    devices = {
	                'dombus - (1.4) EV Mode',
	                'Solax - Remote Control Mode'
	               }
	},
	execute = function(domoticz, Item)
        local evmode = domoticz.devices('dombus - (1.4) EV Mode') 
        local rcmode = domoticz.devices('Solax - Control - Remote Control Mode')
        local rcbutton = domoticz.devices('Solax - Control - Remote Control Trigger') 
        -- Check whether the EV mode is "OFF or Solar" or "from grid"
        if evmode.levelName == 'Off' or evmode.levelName == 'Solar' then
            -- Allow battery to discharge
            rcmode.switchSelector('Disabled')
            rcbutton.switchOn()
        else
            -- Disable battery discharge
            rcmode.switchSelector('Charge Only')
            rcbutton.switchOn()
        end
	end
}
````
For the future me (and others whom are interested :-D), I am using other two scripts to start and stop charging of the EV at a precise time, and my charging system is a dombusEVSE (its devices are also shown below)
[image]

Here is a charging history, from 2AM to 4:30PM at 4.5kW
[image]

closed 2024-10-15 by averter