saidlm/Domoticz-Solax-plugin #13: How to update T1/T2 and R1/R2 of "Total Grid Energy" device

averter, 2024-11-16:
I'm  trying to use the "Total Grid Energy" device to correctly calculate energy costs. 

Similarly to what is described in the [domoticz forum](https://forum.domoticz.com/viewtopic.php?p=321353#p321353) I have two tariffs during each day, which in domoticz are apparently defined as T1, R1 and T2, R2 for usage and return, and this distinction needs to be sent directly to the device. 

So my question is if there is a way to set the times of the day during which the "Total Grid Energy" device controlled by this plugin will be in tariff 1 and tariff 2, so that energy prices can be calculated correctly? As far as I can tell the plugin sends everything into T1 and R1? (screenshot below)
[image]

Thanks for any pointers/help, and once again excellent work in developing this plugin.

saidlm, 2024-11-20:
Hi,
there is no configuration option for that. Honestly speaking I have no plan to add it just because it is not the function of the inverter.  Unfortunately there is no possibility to read two tariffs information from meter connected and read by inverter. Correct me if I am wrong.

There are (at least) two possibility how to handle it.

1. Write short LUA/DzVents event script which will update new virtual device based on time and information from the inverter.
2. Use LUA/DzVents event script which will read the state of switch which indicates low/high tariff and base on that will feed new virtual device providing energy consumption information similar like in previous solution.  The status switch can be virtual one with time plan or real sensors which will read real the information from grid (such as Shelly module or similar one properly connected to command wire).

Martin

saidlm, 2024-11-20:
Here is the example of the scrip from message above:

`return {
        on = {
                devices = {'Consumption Total'}
        },
        execute = function(domoticz, device)
                totalPower = domoticz.devices('Consumption Total').actualWatt;

                if (domoticz.devices('Tariff').state) == "On" then
                    domoticz.devices('Consumption T1').updateElectricity(totalPower);
                    domoticz.devices('Consumption T2').updateElectricity(0);
                else
                    domoticz.devices('Consumption T1').updateElectricity(totalPower);
                    domoticz.devices('Consumption T2').updateElectricity(0);
                end
        end
}`

M.

averter, 2024-11-20:
Hi Martin,
My plan was 1) to put the information regarding prices in the settings "meters" tab
[image]
2) to change the following line of the code (see the location of my cursor in the figure below) which I think is the one that updates the total grid energy device, so that it changes from T1/R1 to T2/R2 based on two new variables which are the time of day at which the tariff 2 starts and ends, e.g., call it STIME2 and ETIME2
[image]
and 3) these two variables could then be added to the already existent settings of the plugin, present in the domoticz hardware interface (below)
[image]
Does this makes sense?

saidlm, 2024-11-20:
Hi,
well, yes it make sense and it will work (most probably), I think. But ... My original plan was to avoid any let me say added value functions such as automation etc. just create interface to Inverter and let everything else up to end user.

M.

averter, 2024-11-21:
Thanks for your thoughts. I agree with that strategy regarding automations being left external to the plugin, and it's what I use to charge my EV for example, but this one is so simple that I wouldn't even call it an automation (it's probably just an extra if-else statement)? 
It could also be implemented in a separate branch and if there's more users interested in this feature it could then be merged? Feel free to close the issue if you prefer to leave it as is.

saidlm, 2024-11-21:
Ok. I understand your point. I spent some time to thing about that and I think I have found better solution how to fulfill the requirement.

My idea is to add the next device - switch in to the plugin device collection. It will be use just for the switching of the tariff of energy meter from outside of the plugin. There are several reason for that.
1. There can be different number of interval for low price tariff during the day. It is not so easy to cover it in plugin config. As soon as the switching is driven by device standard time table can be used.
2. Time table can be easily changeable a nicely visible. Hopefully there is also way how to automate the changing of it (I have no experience with it yet). If there if it can be done base on website or any other external source.
3. Time table is not only solution how to change tariff. You can also use sensor such as Shelly or it can be easily driven by some automation script etc.

What do you think about described solution?

Martin

averter, 2024-11-23:
Hi Martin, 
Yes, I think that in domoticz that is what is called "dynamic price calculation", and yes, it gives more flexibility than a simple low/high tariff. I've also seen plugins to get instantaneous prices into domoticz e.g. [Enever](https://wiki.domoticz.com/Enever).
So a script that would be suitable for my case would have to

1. activate when in the low tariff (2) time period AND the total grid energy device is changed
2. save the power/energy that was sent to the device in a variable, and check if it is usage or return
3. update the total grid energy device with the power/energy in T1/R1 set to 0 and use the variable saved in point 2 to update T2/R2

Something like the above pseudocode? 
The only point that I am not sure about is if I can replace the values that were originally in T1/R1 with zero, after having saved them in a separate variable? In essence, I want to make sure that T1 and R1 are zero during the entire low tariff period.

averter, 2024-11-23:
I think that this might be it (had to use the `Grid Power` device to get power information). Any thoughts?

````lua
return {
	on = {
		devices = {
            ['Solax - Total Grid Energy'] = {'between 00:30 and 05:30'},
		}
	},
    execute = function(domoticz, triggeredItem)
        local TGE = domoticz.devices('Solax - Total Grid Energy')  
        local GP = domoticz.devices('Solax - Grid Power')
        --print(GP.actualWatt)
        if GP.actualWatt < 0 then
            cons = abs(GP.actualWatt)
            prod = 0
        else
            cons = 0
            prod = GP.actualWatt
        end     
        U1 = TGE.usage1
        R1 = TGE.return1
        TGE.updateP1(0, U1, 0, R1, cons, prod).silent()
	end
}
````

averter, 2024-11-24:
I'm very happy with the following script, which does exactly what I want. However, it's necessary to create two integer user variables `t_usage` and `t_return` which need to be set to 0 at the beginning. I've tested it on a mock up dummy P1 device and it worked really well. In this way the total grid energy will calculate the correct allocations of energy and costs for the high and low tariffs. Happy to close it now
```` lua
return {
	on = {
            timer = {'at 00:30','at 05:30'},
            devices = {'Solax - Total Grid Energy'},
	},
    execute = function(dz, item)
        local TGE = dz.devices('Solax - Total Grid Energy')  
        local GP = dz.devices('Solax - Grid Power')
        -- Update U/R energy when at tariff change point
        if item.isTimer then
            if dz.time.matchesRule('at 00:30') then -- tariff change point
                dz.variables('t_usage').set(TGE.usage1)
                dz.variables('t_return').set(TGE.return1)
            elseif dz.time.matchesRule('at 05:30') then -- tariff change point
                dz.variables('t_usage').set(TGE.usage2)
                dz.variables('t_return').set(TGE.return2)
            end
        end
        -- Update Total Grid Energy device
        if item.isDevice then
            tu = dz.variables('t_usage').value  -- usage value @ tariff change point
            tr = dz.variables('t_return').value  -- usage value @ tariff change point
            -- Compute usage and return
            if dz.time.matchesRule('between 00:30 and 05:30') then -- tariff low
                usage2 = TGE.usage1 - tu
                return2 = TGE.return1 - tr
                usage1 = tu
                return1 = tr
            else                                                   -- tariff high
                usage1 = TGE.usage1 - tu
                return1 = TGE.return1 - tr
                usage2 = tu
                return2 = tr
            end
            -- Compute consumption and production
            if GP.actualWatt >= 0 then -- producing
                prod = GP.actualWatt
                cons = 0
            else                       -- consuming
                prod = 0
                cons = - GP.actualWatt
            end
            -- Update P1 smart meter
            TGE.updateP1(usage1, usage2, return1, return2, cons, prod).silent()
        end
    end
}
````

closed 2024-11-24 by averter

saidlm, 2024-12-03:
Hi,
there is new version of the code on the Git which is include implementation based on additional switch.
There are two new devices which will do what you have asked for. The first one is the next smart meter which will show energy based on tariff information. The second one is the switch which can handle tariff changes.  
You can use the switch as you want - hange tariff manually, use another device (sensor) to trigger it, use event script or use timetable of the switch to configure time plan.

Please try it and let me know if it works as expected.

Martin

averter, 2024-12-03:
WoW. It looks perfect :-D! Thank you very much.
So based on the new code, if I want to define a low tariff (T2/R2) between 00:30-05:30, just need to turn the switch "on" at 00:30 and "off" at 05:30 using the switch timers. Like so
[image]